Luos-io/luos_engine #36: Type and Id liste compression to optimize research

nicolas-rabault, 2020-09-28:
Currently during the reception of a message we have to loop on module table to find if we have a module concerned by this messages. This loop take time and have to be executed in IRQ...
To save some time defining if this message is for us or not we could check it by using masks on compressed datas of hosted modules in the node.

To do that the Id or Type value represent a bit decay on a specific buffer containing compressed datas of modules hosted in a node. To check if a message is for us we just have to decay a bit from the ID or TYPE received and compare it with the local compressed values we have.

nicolas-rabault, 2021-11-30:
This is now wat we call filtering...

Salem-Luos, 2022-02-03:
Done on ID.
For type : there is not enough type for now to change the algo.

closed 2022-02-03 by Salem-Luos