Luos-io/luos_engine #72: [BUG] Missing submodules

sfermigier, 2021-02-24:
In the readme: *This repository contains a submodule. The examples folder is linked to another repository. That means this folder won't be cloned with a regular git clone.*

This didn't seem to be the case.

Running `git clone --recursive https://github.com/Luos-io/Luos.git` doesn't make an `examples` directory appear.

nicolas-rabault, 2021-02-24:
Good catch @sfermigier,

Indeed this is something we don't checked from a long time, and I think nobody tried to use it since we make it.

The example folder in the Luos lib is an interesting thing if it is directly usable by IDE (such as platformio) in an examples list menu or something. But our Example repo is more and more complexe and deal with more and more architecture making it complexe to be compatible with this kind of IDE graphical feature.

I think we could remove this submodule.

@sfermigier, do you have any insight about it?

sfermigier, 2021-02-24:
"do you have any insight about it?" -> Nope, just trying to figure out the organisation of your repositories.

nicolas-rabault, 2021-02-24:
ok, so I will remove it.

commit 87fe184d (2021-02-24): Remove unused Examples submodule, fix #72
commit 95c93f51 (2021-02-24): Merge pull request #73 from Luos-io/submodules

Remove unused Examples submodule, fix #72
closed 2021-02-24 by Emanuel-Allely