Luos-io/luos_engine #117: [BUG] Pyluos is stuck when there is a bad behaviour of luos detection

JeromeGalan, 2021-11-08:
**Describe the bug**
When connecting to a Gate, if Pyluos doesn't receive any detection from the Gate : the process is stucked in serial initialisation method. A bypass way is to disconnect and reconnect manually the serial link (usb/serial link in my case) But it's not convenient. A software fix should be usefull.

**Network configuration**
- Bad configuration : a MCU in Luos Network is badly initialised (so the PTP states are totally crashed).

**How to reproduce the bug**
With a network with a bad firmware loaded (specially with bad PTP configurations) , the Gate is logically not able to receive a routing table.

**Additional context**
Even if the bug is caused by a bad user configuration, Pyluos should include a way to properly close serial link (with a software command ? A timeout ? ...).

K0rdan, 2021-11-09:
I think i reproduced this behavior many times by the past with the Typescript SDK

JeromeGalan, 2022-04-28:
#137
https://github.com/Luos-io/Pyluos/pull/137

nicolas-rabault, 2022-05-11:
This have been fixed

closed 2022-05-11 by nicolas-rabault