Luos-io/luos_engine #150: [NEW FEATURE]

fleborne, 2022-02-04:
**What feature would you like to be included in Luos?**

This issue could be related to https://github.com/Luos-io/LuosHAL/issues/40

Just before flashing a new firmware through the bootloader, the Flash is erased. In the current implementation, the whole Flash after `APP_ADDRESS` is erased, whereas only the pages needed to fit the new firmware could be erased.

Because of this behaviour, the user can't use the Flash pages located after their app to store values. Only the shared memory seems to remain unerased.

ncros3, 2022-02-04:
Hello @fleborne , you're right and I will fix this. I think we can close this issue and only keep [issue 40](https://github.com/Luos-io/LuosHAL/issues/40) because the problem is located in the luos HAL.

JeromeGalan, 2022-06-29:
Done in luos engine 2.5.0

closed 2022-06-29 by JeromeGalan