Luos-io/luos_engine #235: [BUG]

nicolas-rabault, 2022-05-20:
**Which version the bug has been detected**
Luos engine 2.3.2

**Describe the bug**
I'm working on a critical application that need 100% message to be sent and not 99% or 101%. In this setup I see some really rare messages received and transmitted multiple time.
In this specific case I spotted a message acked but retransmitted anyway : 
[image]

**Network configuration**
at least 2 nodes.

**How to reproduce the bug**
I have a data and I send the derivative form of this data. On the reception I compute the integral of this data after few time you can compare your data to see if there is any error.

nicolas-rabault, 2022-05-20:
Sometime, The ack doesn't generate any IRQ.
Im using L0...

[image]

nicolas-rabault, 2022-05-20:
When IRQ are disabled at the end of ack, we don't get the ACK.
[image]

JeromeGalan, 2022-11-08:
Linked to #385

Salem-Tho, 2023-01-05:
> Linked to #385

#385 is another bug cause by this one

Salem-Tho, 2023-01-05:
fix in 2.8.0

closed 2023-01-05 by Salem-Tho