Luos-io/luos_engine #237: [BUG] Bootloader cannot reflash after the flash of a gate

BiliouriV, 2022-05-25:
** Which version the bug has been detected **
Luos engine 2.4.2

**Describe the bug**
Each time we flash a gate with the bootloader, so finally we have 2 gates and 2 pipes, and we want to reflash the node, the node is not responding, so we cannot reflash. However the network continues working, if we launch a detection we still have the 2 gates and 2 pipes

**Network configuration**
[image]

and we flash with a gate binary.

**How to reproduce the bug**
Flash 1 time with a gate binary, and then try to reflash with any binary.

**Additional context**
Add any other context about the problem here.
Don't hesitate to add screenshots or to link an entry from the forum.
[image]

BiliouriV, 2022-06-28:
This was a Pyluos bug and it is resolved here: https://github.com/Luos-io/Pyluos/pull/144

JeromeGalan, 2022-07-04:
Correctet in Pyluos release 2.2.2

closed 2022-07-04 by JeromeGalan