Luos-io/luos_engine #251: [BUG]  Code quality :  cleaning

JeromeGalan, 2022-06-17:
**Describe the bug**
The code need a little "summer cleaning" to increase its quality :

- Header functions: fix headers
- Delete old functions
- Warning :  decrease the compilation warning number  / add casts if needed
- Comments  :  check comments + add new comments
- To be completed...

JeromeGalan, 2022-06-21:
Add : clean unit tests (timestamp, .etc..)

BiliouriV, 2022-07-19:
@JeromeGalan I will just leave a comment here so that we do not forget to add the parameter MAX_RTB_ENTRY in the comments of node_config.h because it is a parameter that user could need to modify depending on their needs.

BiliouriV, 2022-07-20:
NOT TO FORGET! In the boot app examples of G431 I forgot #ifdef BOOTLOADER_MODE in main.c and st files. It does not change anything, neither creates a bug. To remove.

BiliouriV, 2022-08-01:
Also, we need to move the INIT_TIME from node_config.h to gate_config.h so that we dont have to define it in every gate project, and add only a comment to the node_config.h

JeromeGalan, 2022-08-18:
Baudrate & luos_flush : should we remove these functions ?

closed 2022-08-31 by JeromeGalan