Luos-io/luos_engine #292: [NEW FEATURE] Native support

nicolas-rabault, 2022-07-26:
# Why
Luos_engine is made to work in a bare metal embedded environment. So, to get started with this technology, you will need at least a computer and a supported dev board.
Find a board and plug it can be a drop condition for a new user.

The `Native support` goal is to allow people to test the technology without needing anything other than a computer.
This issue will be considered closed when [all the training tutorials](https://www.luos.io/tutorials?category=trainings) will be completely compatible with a Native target.

# Constraints
The way things work has to be as close as it is when we use it on embedded targets.
We will have to keep the Luos_engine code base clean. No exception on this part of the code will be allowed. The only place where we can do native-specific things is on HAL and service code.
Here is a Global representation of Luos_engine layers : 
[image]

Creating a Luos engine Core HAL will be straightforward. We only need a systick and can ignore Flash management for now because the bootloader is not part of the training tutorial yet.
Today, there is only one available network called Robus. This network has been thought to work on a half-duplex serial network, so we will have to create a native HAL for Robus to emulate this serial connection through TCP/IP.
We will have to create a specific native HAL (in green) like that : 
[image]

Luos define a `node` as an MCU on the network. In the native target, we could have multiple binaries running on the same CPU; each running binary will be considered a node.

# Used protocols
## Data exchange
Robus rely on multipoint broadcast communication. Every message sent will be received by every node on the network simultaneously (by neglecting propagation time). We will have to emulate this by broadcasting information to every service available on the network.
We will use the WebSocket protocol to send and receive information. WebSocket will allow us to be future-proof and to make things run on web interfaces.

## Discovering
Luos also need to be able to detect other nodes on the network. Robus use PTP lines to manage and find the topology. Those PTP lies allow us to daisy chain the nodes and to know which node is connected to which node, giving us a deterministic topology of the system.
Because native target nodes are binary and not physical parts, we can't use any physical specificities to create a deterministic topology. We will have to find another way.
For now, Luos and Robus are still highly coupled, and the way Luos works could be a problem for this detection feature. We will probably have to do some dirty hacks to make it work. This is ok only if we keep it on services and HAL. 
Mdns is massively used to declare a service on a TCP/IP network. We could use that.
Pear to Pear libraries seems to be able to discover pears and create a list sorted by "distance". This list could be deterministic... Those libs also deal with pear data exchange.

# Development steps
Before a complete setup, we must be able to compile a minimal Luos_engine binary working on a native target.
## Step 1:
So the first step will be to compile an example only using localhost (internal to the binary) communication.
This step has its issue with more details: #290 
## Step 2:
Then we can actually work on developing a native HAL able to communicate on the Robus HAL.
This step has its issue with more details: #291

AltayAkkus, 2022-07-26:
# Thoughts:

> Pear to Pear libraries seems to be able to discover pears and create a list sorted by "distance". This list could be deterministic... Those libs also deal with pear data exchange.

First off: Its "Peer to Peer", a pear is this :D 
[image]

In embedded Luos we have the following topology:

[image]

WebSockets are similiar to USART, it is used for (full) duplex point to point communication.

This is the lifecycle of a WebSocket connection
[image]

So because each Point to Point connection needs a session, there are two possible ways to solve this:

## First option:
[image]
In this setup, each node (except the last node, n-1) would need to host a WebSocket Server, so that Bob can connect to  Alice's server, and they communicate between each other.
Bob then hosts a server, so Charlie can connect to it, so and so forth.
This would, imo, create a huge overhead that is not really needed.
### Problems:

- How does one node know, which node it wants to connect to? In embedded we have this sorted via the Layer 8, but how is it gonna happen here?
- Each node except one needs to host a WebSocket server in order to achieve the PTP connection wanted.
  - While in embedded applications this is necessary due to USART not being a Bus, this is unnecessary overhead for each and every of the binaries/nodes in native.

## Second option
[image]

All nodes have a library to interact with a WebSocket server, yet they don't have to become one.
There is a "master" Native WebSocket Server, and all nodes are his slaves.
The Master Server simply accepts all messages in all sessions (one session per node) and broadcasts them to all the other sessions (nodes).
### Problems
- How does one node know, which node it wants to connect to? In embedded we have this sorted via the Layer 8, but how is it gonna happen here?
  - We may just need to give up routing in this case. In order to be compatible with all current tools and diagnostics we could just fake a routing table via the Master Server, the first node to connect is the first, the second the second, the first is connected to the second, the second to the third et cetera. This would create the illusion that in order for Alice to talk to Charlie, its message gets passed over Bob, but in reality every node is one hop away from every node. 
  - This could be implemented by the master server responding to the [detect message](https://github.com/Luos-io/luos_engine/blob/a7ec2b01818a78c99dfb32c291bdae1c60e754b4/engine/core/src/luos_engine.c#L1220-L1241) with the fake routing table, and not broadcasting it further.
- Each node needs to implement a library that can communicate with WebSocket servers.
  - We could solve this by adding a native HAL, which simply configures all the standard Luos API interfaces for sending and receiving to be made over our WebSocket server, and not via USART or similiar pins on STM32, ESP32, Arduino et cetera. I don't know (yet) if the HAL templates are able to do so.
- We need to implement features like node_detected
  - I think we can successfully fool [this](https://github.com/Luos-io/luos_engine/blob/179f0b7f4bf72bf75f6a77748068bfba5dff5368/network/robus/src/robus.c#L410-L489) into eating our fake routing table, since this apparently sets the node_detected state. Should be no problemo, I think 😄 

Anyway, enough from me for today, I am heading off to my local hackerspace 🍻

nicolas-rabault, 2022-07-27:
`First off: Its "Peer to Peer", a pear is this :D` Thank you for this precision 🤣

In reality embedded Luos network looks like this : 
[image]
It is using Half duplex serial and PTP are simple Gpio.
I agree with you, the second solution seems to be a good one. This is how ROS 1 is working.
In ROS2 they successfully remove the broker to have something closer to Luos without any master.
That's why I think Mdns could be good.
Mdns is able to create a list of a specific applications available on the network and gives an address to send them things. I think we can even directly use it to broadcast messages to every available service (But I think it's a hack of the tech).
Because of the point to point aspect of the WebSocket I'm not so sure we will be able to use it finally.
Perhaps @mjansson could help us...

nicolas-rabault, 2022-07-27:
I talked with @mjansson on Discord few days ago and he tell me : 
 ```
mDNS in general is for querying specific information, like finding the IP address of a specific node after doing a DNS-SD discovery. Using mDNS to broadcast data is not really what it was intended for, altough I guess you could do it by appending additional TXT records to your annouce/answers.
I would suggest either using mDNS & DNS-SD to find all the instances on the subnet, then unicast packets to each node.  Or use multicast UDP packets on some other port to send packets to all hosts (which quickly turns bad if the number of nodes are large)
```
In our case we are using Serial on normal condition I don't think UDP could be a problem for now.

AltayAkkus, 2022-07-27:
The thing is that WebSockets are not a virtual network. There is no topology like a network.

You have a server, and you have clients. Each client is isolated, and it talks to you in a single session, that session has 2 participants, server and client, and no more.

There is no need to use things like mDNS, or unicast packets to each node, it's like generating a list of users from a database, you don't need service discovery for that either, you just need to read the table and output it.

The WebSocket Master would not act as a master in traditional sense, like that it has higher privilege or priority, it is just a tool to make the intercom of the Nodes possible.

Also: The nodes don't have IP adresses allocated. They communicate via Port 80 (HTTP) or 443 (HTTPS), they don't have a ip adress for identification, the identification is handled in the session.

closed 2023-03-31 by JeromeGalan