Luos-io/luos_engine #295: [NEW FEATURE] Secure OD structure usage avoiding users to directly set it

nicolas-rabault, 2022-07-29:
Object Dictionary are important to keep your application loosly coupled, see : https://www.luos.io/blog/why-is-standardized-data-so-important-in-embedded-systems
see documentation : https://www.luos.io/docs/luos-technology/message/object-dictionary
see concerned folder : https://github.com/Luos-io/luos_engine/tree/main/engine/OD

To make Object Dictionary useful we need to be sure that users never directly write on OD (Object Dictionary) variables. If they do that the code become unsafe because the value will be stored with an unknown unit(or only known by the developer). OD structures should be always written using the dedicated functions forcing users to specify the used unit.

To do that I propose to modify the OD structure by adding a layer with a `reserved` var.
Fort example, instead of : 
```
typedef float angular_position_t;
```
We should do:
```
typedef struct{
    float reserved;
} angular_position_t;
```

This way, to access the value, users have to do `My_var.reserved = 10`. This should convince them not to and to use functions instead.

All the functions of the OD will be able to use the `reserved` var to write or return a converted value.

Faizan711, 2022-11-28:
Hey guys, is this issue still open, if so I would like to be assigned for it.

nicolas-rabault, 2022-11-29:
Hi @Faizan711,
Yes this Issue is still open! Thank you
@Simonbdy just assigned you.
If you have any questions please let me know here or on Discord.

Faizan711, 2022-11-30:
sure i will connect with you on discord.

Faizan711, 2022-12-02:
`/******************************************************************************
 * @file OD_angular
 * @brief object dictionnary angular
 * @author Luos
 * @version 0.0.0
 ******************************************************************************/
#ifndef OD_OD_ANGULAR_H_
#define OD_OD_ANGULAR_H_

#include <string.h>
/*******************************************************************************
 * Definitions
 ******************************************************************************/
typedef struct{
    float private;
} angular_position_t;

/*******************************************************************************
 * Variables
 ******************************************************************************/

/*******************************************************************************
 * Function
 ******************************************************************************/
// angular_position are stored in degree (deg)
//******** Conversions ***********

// deg
static inline float AngularOD_PositionTo_deg(angular_position_t self)
{
    return self.private;
}

static inline angular_position_t AngularOD_PositionFrom_deg(float deg)
{
    return deg;
}

// rev
static inline float AngularOD_PositionTo_rev(angular_position_t self)
{
    return self.private / 360.0f;
}

static inline angular_position_t AngularOD_PositionFrom_rev(float rev)
{
    return rev * 360.0f;
}

// rad
static inline float AngularOD_PositionTo_rad(angular_position_t self)
{
    return (self.private * 2.0f * 3.141592653589793f) / 360.0f;
}

static inline angular_position_t AngularOD_PositionFrom_rad(float rad)
{
    return (rad * 360.0f) / (2.0f * 3.141592653589793f);
}

//******** Messages management ***********
static inline void AngularOD_PositionToMsg(const angular_position_t *const self, msg_t *const msg)
{
    msg->header.cmd = ANGULAR_POSITION;
    memcpy(msg->data, self.private, sizeof(angular_position_t));
    msg->header.size = sizeof(angular_position_t);
}

static inline void AngularOD_PositionFromMsg(angular_position_t *const self, const msg_t *const msg)
{
    memcpy(self.private, msg->data, msg->header.size);
}

typedef struct{
    float private;
} angular_speed_t;

// angular_speed are stored in degree/s (deg/s)
//******** Conversions ***********

// deg_s
static inline float AngularOD_SpeedTo_deg_s(angular_speed_t .privateself)
{
    return self.private;
}

static inline angular_speed_t AngularOD_SpeedFrom_deg_s(float deg)
{
    return deg;
}

// rev_s
static inline float AngularOD_SpeedTo_rev_s(angular_speed_t self)
{
    return self.private / 360.0f;
}

static inline angular_speed_t AngularOD_SpeedFrom_rev_s(float rev_s)
{
    return rev_s * 360.0f;
}

// rev_min
static inline float AngularOD_SpeedTo_rev_min(angular_speed_t self)
{
    return self.private * 60.0f / 360.0f;
}

static inline angular_speed_t AngularOD_SpeedFrom_rev_min(float rev_min)
{
    return rev_min * 360.0f / 60.0f;
}

// rad_s
static inline float AngularOD_SpeedTo_rad_s(angular_speed_t self)
{
    return (self.private * 2.0f * 3.141592653589793f) / 360.0f;
}

static inline angular_speed_t AngularOD_SpeedFrom_rad_s(float rad_s)
{
    return (rad_s * 360.0f) / (2.0f * 3.141592653589793f);
}

//******** Messages management ***********
static inline void AngularOD_SpeedToMsg(const angular_speed_t *const self, msg_t *const msg)
{
    msg->header.cmd = ANGULAR_SPEED;
    memcpy(msg->data, self.private, sizeof(angular_speed_t));
    msg->header.size = sizeof(angular_speed_t);
}

static inline void AngularOD_SpeedFromMsg(angular_speed_t *const self, const msg_t *const msg)
{
    memcpy(self.private, msg->data, msg->header.size);
}

#endif /* OD_OD_ANGULAR_H_ */`

Faizan711, 2022-12-02:
@nicolas-rabault  can you please check the given change, if it is correct i will apply it to all files. If there is something incorrect please let me know that too i will try to get it correct.

nicolas-rabault, 2022-12-03:
Sorry about the delay I was traveling.
To simplify review you should create a commit with your code. This allow me to see the differences between your version and the original one. Also a bot will try to compile for you and report you most of the issues you could have on your code.
Don't worry commits here can be work in progress, and no commit is definitive, we ca reshape it in the end.
Just by looking to your code perhaps there is some cast missing where functions return is `angular_speed_t`. Alos there is some indentation mistakes but this is perhaps du to your copy past on the GH comment.
Please put it on a commit.
To do it, you have to fork this repo on your own repo, then create a branch, add your commit, then create a Pull Request to our repo and mention this issue on your PR.

Faizan711, 2022-12-03:
I have made the PR. Thanks a lot for your guidance. Please let me know the review of it.