Luos-io/luos_engine #323: [BUG] Some services stop receiving messages from the Gate

fleborne, 2022-08-04:
**Which version the bug has been detected**
Luos engine 2.5.2

**Describe the bug**
After a while (several minutes), random services on the network stop receiving messages from the Gate.

Each PCB features a LED that is lit up for one second after every __command__ message (power_ratio) is received from the Gate. This is why I am sure that the services stop receiving commands from the Gate.

**Network configuration**
It is the same configuration as in https://github.com/Luos-io/luos_engine/issues/308.

The image below shows the network made of one Gate + Pipe (NUCLEO-G431KB with custom shield) and 25 custom nodes (STM32G431KB based), each one hosting one custom ServoMotor service.

The Gate + Pipe are powered through the USB cable. Each motor control node is powered (12 V regulated down to 3.3 V by each PCB) through the harness visible in the middle of the image (U shape).

[image]

The list of nodes and services is the following:

```
  ┏━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┓
  ┃  ╭node 1            /!\ Not certified            ┃
  ┃  │  Type                Alias               ID   ┃
  ┃  ├> Gate                gate                1    ┃
  ┃  ╰> Pipe                Pipe                2    ┃
╔>┗━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┛
║     ┏━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┓
╚══ 1>┃0 ╭node 2            /!\ Not certified            ┃
      ┃  │  Type                Alias               ID   ┃
      ┃  ╰> ServoMotor          nb55                3    ┃
    ╔>┗━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┛
    ║     ┏━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┓
    ╚══ 1>┃0 ╭node 3            /!\ Not certified            ┃
          ┃  │  Type                Alias               ID   ┃
          ┃  ╰> ServoMotor          nb551               4    ┃
        ╔>┗━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┛
        ║     ┏━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┓
        ╚══ 1>┃0 ╭node 4            /!\ Not certified            ┃
              ┃  │  Type                Alias               ID   ┃
              ┃  ╰> ServoMotor          nb552               5    ┃
            ╔>┗━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┛
            ║     ┏━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┓
            ╚══ 1>┃0 ╭node 5            /!\ Not certified            ┃
                  ┃  │  Type                Alias               ID   ┃
                  ┃  ╰> ServoMotor          nb553               6    ┃
                ╔>┗━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┛
                ║     ┏━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┓
                ╚══ 1>┃0 ╭node 6            /!\ Not certified            ┃
                      ┃  │  Type                Alias               ID   ┃
                      ┃  ╰> ServoMotor          nb554               7    ┃
                    ╔>┗━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┛
                    ║     ┏━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┓
                    ╚══ 1>┃0 ╭node 7            /!\ Not certified            ┃
                          ┃  │  Type                Alias               ID   ┃
                          ┃  ╰> ServoMotor          nb555               8    ┃
                        ╔>┗━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┛
                        ║     ┏━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┓
                        ╚══ 1>┃0 ╭node 8            /!\ Not certified            ┃
                              ┃  │  Type                Alias               ID   ┃
                              ┃  ╰> ServoMotor          nb556               9    ┃
                            ╔>┗━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┛
                            ║     ┏━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┓
                            ╚══ 1>┃0 ╭node 9            /!\ Not certified            ┃
                                  ┃  │  Type                Alias               ID   ┃
                                  ┃  ╰> ServoMotor          nb557               10   ┃
                                ╔>┗━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┛
                                ║     ┏━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┓
                                ╚══ 1>┃0 ╭node 10            /!\ Not certified            ┃
                                      ┃  │  Type                Alias               ID   ┃
                                      ┃  ╰> ServoMotor          nb558               11   ┃
[... 93 lines cut ...]

#define TX_EN_PIN GPIO_PIN_15
#define TX_EN_PORT GPIOA

#define MAX_RTB_ENTRY            80
#define GATE_BUFF_SIZE           2048
#define PIPE_TO_LUOS_BUFFER_SIZE 2048
#define LUOS_TO_PIPE_BUFFER_SIZE 4096

#define INIT_TIME 150

#endif /* _NODE_CONFIG_H_ */
```

**How to reproduce the bug**
1. Set update_time of all servomotor services to 0.075 s (~13.33 Hz)
2. Configure all servomotor services in power mode with only rot_position as feedback
3. Send a power_ratio command to each servomotor, with a pause of 1 ms between each and a longer pause after the last servomotor of the network so that the update frequency matches ~13.33 Hz

**Additional context**

-  [network statistics at the end of the test](https://github.com/Luos-io/luos_engine/files/9261650/luos_stats_12minutes_25moteurs_13Hz.csv)
- monitoring the refresh_freq attribute of each PyLuos object representing a servomotor service, it appears that PyLuos continues to receive data from the Gate for all services, because all refresh_freq continue to change slightly.

In another test with only 6 servomotor services, all services except the first one stopped receiving data after 854 seconds of testing. After restarting the test without rebooting the system, the last 5 services stop receiving data after a couple seconds. Rebooting the Gate changed nothing, but rebooting the servomotor services allowed the system to work for several minutes before encountering the same issue.

fleborne, 2022-08-08:
Further investigations have shown that this problem occurs when we send a command that is constant for each service.

In my case, the following examples:

```python
# changing command for all motors: works
for mot in device.services[2:]:
  mot.power_ratio = 100.0 * math.sin(time.time())

# same constant command for all motors: doesn't work
for mot in device.services[2:]:
  mot.power_ratio = 100.0

# different constant commands for each motor: doesn't work
for i, mot in enumerate(device.services[2:]):
  mot.power_ratio = float(i)
```

JeromeGalan, 2022-09-27:
Fixed in luos_engine 2.6.3

closed 2022-09-27 by JeromeGalan