Luos-io/luos_engine #327: [BUG] Services are excluded in case of NBR_RETRY exceed only in case of collision.

nicolas-rabault, 2022-08-10:
** Which version the bug has been detected **
Luos engine 2.5.2

**Describe the bug**
Luos_engine should exclude services in case of communication trouble. But in fact it seems to not exclude services that doesn't reply.
This is coming from this line : https://github.com/Luos-io/luos_engine/blob/main/network/robus/src/transmission.c#L131

**How to reproduce the bug**
Use a Gate and an other node, send data to the node from the gate then unplug the node. Gate should send you an alert due to services exclusion, but it's not...

fleborne, 2022-08-10:
Removing the `if (ctx.tx.collision)` test line 131 solves the issue.

I don't know about potential side effects though

nicolas-rabault, 2022-08-10:
I don't see any!
The system will just be less permissive...

JeromeGalan, 2022-10-26:
We will remove this "if" statement.

In a future release, we should only excluded a node if no ACK and/or NACK has been received.
If there are collisions, the other node is not crashed in all cases: maybe it can answer later or the network was oveloaded.

JeromeGalan, 2022-10-26:
This last comment is described here #392 
Current issue 327 will be closed when 2.7.0 will be released

JeromeGalan, 2022-11-04:
Done in Luos engine 2.7.0

closed 2022-11-04 by JeromeGalan