Luos-io/luos_engine #385: [BUG] : failling detection process due to ack miss understanding

Salem-Tho, 2022-10-12:
When a local routing is ask the ack of this message is not catch so the message is resend! Pb is that the node that need to send local routing understand the two time the message ask the routing table. so the local routing table is send twice.

During the process of sending the global routing table, the local rtb sended twice is pull only one time and stuck in the buffer as oldest message and when the buffer is fill with the global rtb message quand go out and an assert occur

'''
        // Send message
        uint32_t tickstart = Luos_GetSystick();
        while (Luos_SendMsg(service, msg) == FAILED)
        {
            // No more memory space available
            // 500ms of timeout after start trying to load our data in memory. Perhaps the buffer is full of RX messages try to increate the buffer size.
            LUOS_ASSERT(((volatile uint32_t)Luos_GetSystick() - tickstart) < 500);
        }
'''

it can be reproduce on stairs project when a big rtb is send and a local routing table is received twice

JeromeGalan, 2022-10-17:
[TODO]  This issue could be reproducted in a unit test scenario

Salem-Tho, 2023-01-05:
need to create a  unit test with a msg stuck in the message buffer

JeromeGalan, 2023-02-03:
Done in 2.8.0

closed 2023-02-03 by JeromeGalan