Luos-io/luos_engine #391: [BUG] PTP stucked to high level during booltloader data transfer

JeromeGalan, 2022-10-20:
## Details
Please replace the italic test by your inputs and details in each of the following sections:

### Which version the bug has been detected on
*Luos engine 2.6.4*

### Description of the bug
The PTP is stucked to HIGH level after a 2nd flash with pyluos bootloader.

[image]

### Context and environment
Gate L0 + Nucleo L432KC with a LED service

### How to reproduce the bug
The Nucleo was in bootloader mode. A Led service has been flashed with pyluos bootloader. We try to flash it a second time with the same service.

### Possible solution
Maybe we should do a "pyluos-bootloader reset" before each bootloader flash ? Or even call a reset automatically in bootloader machine state ?
To be verified : is it specified in the documentation ?

JeromeGalan, 2022-10-20:
The salae trace is on my PC (Desktop\Timeline\Issue_391.sal)
Ask it on demands!

nicolas-rabault, 2023-11-13:
Can't see this anymore.

closed 2023-11-13 by nicolas-rabault