Luos-io/luos_engine #457: Attiny1616 or attiny3216 [MCU PORTING]

concon, 2023-12-18:
Hello,

I want to use Luos with Attiny1616 or 3216, but I don't have enough knowledge about Hal. Can you make a HAL study for this?

Kind regards,

nicolas-rabault, 2023-12-19:
Hi @concon,
Thank you for your interest.
I just made a quick check for Attiny1616 or 3216. Attiny1616 will not have enough flash space for you to be comfortable using Luos_engine. I advise you to use Attiny3216.

There are multiple HAL you may need to make depending on the usage of your Attiny3216.

 - The [core Luos_engine HAL](https://github.com/Luos-io/luos_engine/tree/main/engine/HAL) allows you to use Luos_engine. This one mostly needs systick. In your case you probably can't use the bootloader feature due to the lack of flash, so you don't have to take care of the flash management of the HAL. I think the `ATSAMD21_ARDUINO` should be ready to be used on Attiny3216 if you remove the flash management code.

 - There is a HAL for each network you may use. For now only `Serial` and `Robus` are available, but of course, you can create your own. You will have to create HAL for those you plan to use.
 
Unfortunately, no more full-time people are developing this lib, only volunteers. I can help you understand the needs of different HAL but I will not be able to make it by myself.

concon, 2023-12-20:
Thank you very much for your kind and immediate answer, I don't know how to write HAL and what the logic is, but I will look for a solution, by the way, I hope your new situation will bring better conditions for you, it's sad, of course, I wish you could be together.

I wish you continued success in your work,

Best regards,

Lemi

iPad’imden gönderildi

Nicolas Rabault ***@***.***> şunları yazdı (19 Ara 2023 12:44):

Thank you for your interest.
I just made a quick check for Attiny1616 or 3216. Attiny1616 will not have enough flash space for you to be comfortable using Luos_engine. I advise you to use Attiny3216.

There are multiple HAL you may need to make depending on the usage of your Attiny3216.

 - The core Luos_engine HAL<https://github.com/Luos-io/luos_engine/tree/main/engine/HAL> allows you to use Luos_engine. This one mostly needs systick. In your case you probably can't use the bootloader feature due to the lack of flash, so you don't have to take care of the flash management of the HAL. I think the ATSAMD21_ARDUINO should be ready to be used on Attiny3216 if you remove the flash management code.

 - There is a HAL for each network you may use. For now only Serial and Robus are available, but of course, you can create your own. You will have to create HAL for those you plan to use.

Unfortunately, no more full-time people are developing this lib, only volunteers. I can help you understand the needs of different HAL but I will not be able to make it by myself.

nicolas-rabault, 2024-01-03:
Hi @concon 

I didn't write any tutorial to make your own HAL yet. You could make it following the [Documentation related to the Phy creation](https://www.luos.io/docs/luos-technology/phy).
If you have any questions feel free to [reach me directly on the Luos Discord](https://discord.gg/luos-community-902486791658041364)