Luos-io/luos_engine #464: [BUG] ESP32 is constantly resetting for Led example

costa-victor, 2024-01-11:
## Details
Hello there, I've been trying to run the examples for the ESP32 but I'm experiencing some issues. Does anyone faced a similar issue?

### Which version the bug has been detected on
*Luos engine 3.1.0*

### Description of the bug
Led example for ESP32 is building in Luous Engine 3.1.0 but the microcontroller is constantly resetting at the bootup. The serial monitor output can be found below:
```log
␛[0;32mI (57) boot: ESP-IDF 4.4.5 2nd stage bootloader␛[0m
␛[0;32mI (57) boot: compile time 08:25:49␛[0m
␛[0;32mI (58) boot: chip revision: v1.0␛[0m
␛[0;32mI (61) boot.esp32: SPI Speed      : 40MHz␛[0m
␛[0;32mI (66) boot.esp32: SPI Mode       : DIO␛[0m
␛[0;32mI (70) boot.esp32: SPI Flash Size : 4MB␛[0m
␛[0;33mW (75) boot.esp32: PRO CPU has been reset by WDT.␛[0m
␛[0;33mW (80) boot.esp32: WDT reset info: PRO CPU PC=0x400865c0␛[0m
␛[0;33mW (86) boot.esp32: WDT reset info: APP CPU PC=0x400865c6␛[0m
␛[0;32mI (92) boot: Enabling RNG early entropy source...␛[0m
␛[0;32mI (98) boot: Partition Table:␛[0m
␛[0;32mI (101) boot: ## Label            Usage          Type ST Offset   Length␛[0m
␛[0;32mI (109) boot:  0 nvs              WiFi data        01 02 00009000 00006000␛[0m
␛[0;32mI (116) boot:  1 phy_init         RF data          01 01 0000f000 00001000␛[0m
␛[0;32mI (124) boot:  2 factory          factory app      00 00 00010000 00100000␛[0m
␛[0;32mI (131) boot: End of partition table␛[0m
␛[0;32mI (136) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0a810h ( 43024) map␛[0m
␛[0;32mI (159) esp_image: segment 1: paddr=0001a838 vaddr=3ffb0000 size=01bf4h (  7156) load␛[0m
␛[0;32mI (162) esp_image: segment 2: paddr=0001c434 vaddr=40080000 size=03be4h ( 15332) load␛[0m
␛[0;32mI (170) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=1d438h (119864) map␛[0m
␛[0;32mI (213) esp_image: segment 4: paddr=0003d460 vaddr=40083be4 size=0aab0h ( 43696) load␛[0m
␛[0;32mI (238) boot: Loaded app from partition at offset 0x10000␛[0m
␛[0;32mI (238) boot: Disabling RNG early entropy source...␛[0m
␛[0;32mI (249) cpu_start: Pro cpu up.␛[0m
␛[0;32mI (249) cpu_start: Starting app cpu, entry point is 0x40083788␛[0m
␛[0;32mI (0) cpu_start: App cpu up.␛[0m
␛[0;32mI (263) cpu_start: Pro cpu start user code␛[0m
␛[0;32mI (263) cpu_start: cpu freq: 160000000␛[0m
␛[0;32mI (264) cpu_start: Application information:␛[0m
␛[0;32mI (268) cpu_start: Project name:     led␛[0m
␛[0;32mI (273) cpu_start: App version:      3.0.0-9-g73792a51-dirty␛[0m
␛[0;32mI (279) cpu_start: Compile time:     Jan 11 2024 08:25:12␛[0m
␛[0;32mI (285) cpu_start: ELF file SHA256:  a406634436410454...␛[0m
␛[0;32mI (291) cpu_start: ESP-IDF:          4.4.5␛[0m
␛[0;32mI (296) cpu_start: Min chip rev:     v0.0␛[0m
␛[0;32mI (300) cpu_start: Max chip rev:     v3.99 ␛[0m
␛[0;32mI (305) cpu_start: Chip rev:         v1.0␛[0m
␛[0;32mI (310) heap_init: Initializing. RAM available for dynamic allocation:␛[0m
␛[0;32mI (317) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM␛[0m
␛[0;32mI (323) heap_init: At 3FFB3468 len 0002CB98 (178 KiB): DRAM␛[0m
␛[0;32mI (330) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM␛[0m
␛[0;32mI (336) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM␛[0m
␛[0;32mI (342) heap_init: At 4008E694 len 0001196C (70 KiB): IRAM␛[0m
␛[0;32mI (350) spi_flash: detected chip: generic␛[0m
␛[0;32mI (353) spi_flash: flash io: dio␛[0m
␛[0;32mI (358) cpu_start: Starting scheduler on PRO CPU.␛[0m
␛[0;32mI (0) cpu_start: Starting scheduler on APP CPU.␛[0m
␛[0;32mI (368) gpio: GPIO[5]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (378) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (388) gpio: GPIO[26]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (398) gpio: GPIO[26]| InputEn: 0| OutputEn: 1| OpenDrain: 1| Pullup: 0| Pulldown: 1| Intr:0 ␛[0m
␛[0;32mI (408) gpio: GPIO[27]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (418) gpio: GPIO[27]| InputEn: 0| OutputEn: 1| OpenDrain: 1| Pullup: 0| Pulldown: 1| Intr:0 ␛[0m
␛[0;33mW (428) uart: ESP_INTR_FLAG_IRAM flag is set while CONFIG_UART_ISR_IN_IRAM is not enabled, flag updated␛[0m
␛[0;32mI (438) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 ␛[0m
␛[0;32mI (448) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 ␛[0m
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6892
ho 0 tail 12 room 4
load:0x40078000,len:15376
load:0x40080400,len:4000
entry 0x4008068c
␛[0;32mI (57) boot: ESP-IDF 4.4.5 2nd stage bootloader␛[0m
␛[0;32mI (57) boot: compile time 08:25:49␛[0m
␛[0;32mI (58) boot: chip revision: v1.0␛[0m
␛[0;32mI (61) boot.esp32: SPI Speed      : 40MHz␛[0m
␛[0;32mI (66) boot.esp32: SPI Mode       : DIO␛[0m
␛[0;32mI (70) boot.esp32: SPI Flash Size : 4MB␛[0m
␛[0;33mW (75) boot.esp32: PRO CPU has been reset by WDT.␛[0m
```
### How to reproduce the bug
1- Build the Led example within the ESP32 example folder
2- Upload to the ESP32 board
3- Open the Serial Monitor

nicolas-rabault, 2024-01-11:
Hi @costa-victor ,
I'm not very experienced with ESP32
`rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)` seems to indicate a reset due to a watchdog timer or something.
Is the Button example doing the same?

costa-victor, 2024-01-11:
Hello @nicolas-rabault,
Thank you for the quick response.

> Is the Button example doing the same?

Yes, same issue. 

I've just started learning about luos and I've been trying the luos 3.1.0, but I'll try a older version instead just to compare.

costa-victor, 2024-01-11:
Hmm, I see. Thanks for letting me know.

> what kind of network are you planning to use with Luos ?

I'm intending to use a single node but with several services that need to interact with each other.

nicolas-rabault, 2024-01-11:
Interesting!
I think it's the first time someone is trying to do that. I never tried to remove all the network!

costa-victor, 2024-01-12:
Thank you @nicolas-rabault , I did the same testing and it seems to be ok.